earlephilhower/esp-quick-toolchain #5: Building issues

Niek, 2019-07-11:
I encountered some issues trying to build a toolchain for 3.0.0.

https://github.com/earlephilhower/esp-quick-toolchain/blob/99f8e128c2eaf573daac0de7aa2d8d6f478d26f7/Makefile#L258
The site mr511.de no longer seems online, so I had to change the URL to:
https://fossies.org/linux/misc/old/libelf-0.8.13.tar.gz

After that, stage.LINUX.binutils-config is failing:
```
checking build system type... x86_64-pc-linux-gnu
checking host system type... Invalid configuration `linux': machine `linux' not recognized
configure: error: /bin/bash /workdir/repo/binutils-gdb/config.sub linux failed
```

earlephilhower, 2019-07-12:
Thanks, @Niek, for the update.  I never liked having that link to some random guy's repo (sorry, I know he's probably the author but it's a strange place to host things...IIRC even the RH builds pulled from that private server).

I've got the working tarball, I'll add it to the repo and adjust scripts once I get the gcc 7.2 changes merged (it works, amazingly, with almost no changes!).

earlephilhower, 2019-07-12:
What was your exact `make` command line?

I'm starting the whole `make download && make linux` from a completely empty testing directory now.  There were a few hiccups with the GIT repo commands when starting from scratch, but so far it's chugging along OK.  This is on a relatively up to date Ubuntu 18.04 system:

````
earle@server:~/src/test/esp-quick-toolchain$ make download
cat: .ghuser: No such file or directory
cat: .ghpass: No such file or directory
Makefile:23: *** Need to specify GH username on the command line "GHUSER=xxxx" or in .ghuser.  Stop.
earle@server:~/src/test/esp-quick-toolchain$ cp ../../esp-quick-toolchain/.gh* .
earle@server:~/src/test/esp-quick-toolchain$ make download
STAGE: .stage.download
earle@server:~/src/test/esp-quick-toolchain$ make linux -j32
STAGE: .stage.download
STAGE: .clean.binutils-gdb.git
STAGE: .clean.gcc.git
STAGE: .clean.newlib.git
STAGE: .clean.lx106-hal.git
STAGE: .clean.mkspiffs.git
STAGE: .clean.esptool.git
STAGE: .clean.mklittlefs.git
HEAD is now at 24124ef Adjust for cross-build environment
HEAD is now at 7f77f2b Merge pull request #2 from earlephilhower/update
HEAD is now at f80ae31 Update for cross-compiled builds
HEAD is now at e4bcc63 Merge pull request #11 from olaf-mandel/cleanup2
HEAD is now at d230f4322ce xtensa: report stack usage
HEAD is now at 314c6dead Use 1 less register in pgm_read_dword_unaligned
HEAD is now at 2ae0bdddef xtensa: ld: convert tests to run_dump_test
STAGE: .stage.prepgit
STAGE: .stage.checkout
STAGE: .stage.patch
STAGE: .stage.LINUX.start
STAGE: .stage.LINUX.binutils-config
STAGE: .stage.LINUX.mkspiffs
STAGE: .stage.LINUX.esptool
STAGE: .stage.LINUX.mklittlefs
STAGE: .stage.LINUX.binutils-make
STAGE: .stage.LINUX.gcc1-config
STAGE: .stage.LINUX.gcc1-make
STAGE: .stage.LINUX.newlib-config
STAGE: .stage.LINUX.newlib-make
STAGE: .stage.LINUX.libstdcpp
STAGE: .stage.LINUX.libsdtcpp-nox
STAGE: .stage.LINUX.hal-config
STAGE: .stage.LINUX.hal-make
STAGE: .stage.LINUX.strip
STAGE: .stage.LINUX.post
STAGE: .stage.LINUX.package
````

earlephilhower, 2019-07-12:
Oh, and pull the latest git head since I did do the 7.2 merge as well as a few additional tweaks, just a little bit ago.

Niek, 2019-07-13:
Here is my output:
```bash
$ docker run --user $(id -u):$(id -g) --rm -v $(pwd):/workdir earlephilhower/gcc-cross bash -c "cd /workdir; make -j32 GCC=7.2 host=linux GHUSER=earlephilhower GHPASS=x REL=3.0.0 SUBREL=1"
STAGE: .clean.binutils-gdb.git
STAGE: .stage.download
STAGE: .clean.gcc.git
STAGE: .clean.newlib.git
STAGE: .clean.lx106-hal.git
STAGE: .clean.esptool.git
STAGE: .clean.mkspiffs.git
STAGE: .clean.mklittlefs.git
HEAD is now at e4bcc63 Merge pull request #11 from olaf-mandel/cleanup2
HEAD is now at f80ae31 Update for cross-compiled builds
HEAD is now at 7f77f2b Merge pull request #2 from earlephilhower/update
HEAD is now at 7fefeac add license file
HEAD is now at 314c6dead Use 1 less register in pgm_read_dword_unaligned
HEAD is now at 2ae0bdddef xtensa: ld: convert tests to run_dump_test
HEAD is now at d230f4322ce xtensa: report stack usage
Removing cloog-0.18.1/Makefile.in
Removing cloog-0.18.1/aclocal.m4
Removing cloog-0.18.1/autoconf/
Removing cloog-0.18.1/configure
Removing cloog-0.18.1/doc/
Removing cloog-0.18.1/examples/
Removing cloog-0.18.1/isl/Makefile.in
Removing cloog-0.18.1/isl/aclocal.m4
Removing cloog-0.18.1/isl/config.guess
Removing cloog-0.18.1/isl/config.sub
Removing cloog-0.18.1/isl/configure
Removing cloog-0.18.1/isl/depcomp
Removing cloog-0.18.1/isl/doc/
Removing cloog-0.18.1/isl/include/
Removing cloog-0.18.1/isl/install-sh
Removing cloog-0.18.1/isl/interface/
Removing cloog-0.18.1/isl/isl_config.h.in
Removing cloog-0.18.1/isl/ltmain.sh
Removing cloog-0.18.1/isl/m4/
Removing cloog-0.18.1/isl/missing
Removing cloog-0.18.1/m4/
Removing cloog-0.18.1/osl/Makefile.in
Removing cloog-0.18.1/osl/aclocal.m4
Removing cloog-0.18.1/osl/configure
Removing cloog-0.18.1/osl/doc/
Removing cloog-0.18.1/osl/tests/
Removing cloog-0.18.1/test/
STAGE: .stage.prepgit
STAGE: .stage.checkout
STAGE: .stage.patch
STAGE: .stage.LINUX.start
STAGE: .stage.LINUX.binutils-config
STAGE: .stage.LINUX.esptool
STAGE: .stage.LINUX.mkspiffs
STAGE: .stage.LINUX.mklittlefs
Makefile:449: recipe for target '.stage.LINUX.esptool' failed
make: *** [.stage.LINUX.esptool] Error 2
make: *** Waiting for unfinished jobs....
Makefile:432: recipe for target '.stage.LINUX.mklittlefs' failed
make: *** [.stage.LINUX.mklittlefs] Error 2
make: *** [.stage.LINUX.mkspiffs] Error 2
Makefile:415: recipe for target '.stage.LINUX.mkspiffs' failed
Makefile:323: recipe for target '.stage.LINUX.binutils-config' failed
make: *** [.stage.LINUX.binutils-config] Error 1
$ cat log.stage.LINUX.binutils-config
checking build system type... x86_64-pc-linux-gnu
checking host system type... Invalid configuration `linux': machine `linux' not recognized
configure: error: /bin/bash /workdir/repo/binutils-gdb/config.sub linux failed
```

And if I try without docker (Ubuntu 16.04, so outdated):
```bash
$ GHUSER=earlephilhower GHPASS=x make download
STAGE: .stage.download
$ GHUSER=earlephilhower GHPASS=x make linux -j32
STAGE: .stage.download
STAGE: .clean.binutils-gdb.git
STAGE: .clean.gcc.git
STAGE: .clean.newlib.git
STAGE: .clean.lx106-hal.git
STAGE: .clean.mkspiffs.git
STAGE: .clean.esptool.git
STAGE: .clean.mklittlefs.git
HEAD is now at e4bcc63 Merge pull request #11 from olaf-mandel/cleanup2
HEAD is now at 7f77f2b Merge pull request #2 from earlephilhower/update
HEAD is now at 7fefeac add license file
HEAD is now at f80ae31 Update for cross-compiled builds
Removing log.stage.patch
HEAD is now at 314c6de Use 1 less register in pgm_read_dword_unaligned
HEAD is now at 2ae0bdd xtensa: ld: convert tests to run_dump_test
HEAD is now at 3ed0c49 xtensa: fix PR target/82181
Removing cloog
Removing cloog-0.18.1/
Removing libelf
Removing libelf-0.8.13/
STAGE: .stage.prepgit
STAGE: .stage.checkout
Makefile:281: recipe for target '.stage.checkout' failed
make: *** [.stage.checkout] Error 1
$ cat log.stage.checkout
HEAD is now at 3ed0c49 xtensa: fix PR target/82181
Switched to branch 'call0-4.8.2'
Your branch is up-to-date with 'origin/call0-4.8.2'.
HEAD is now at 7fefeac add license file
usage: git submodule [--quiet] add [-b <branch>] [-f|--force] [--name <name>] [--reference <repository>] [--] <repository> [<path>]
   or: git submodule [--quiet] status [--cached] [--recursive] [--] [<path>...]
   or: git submodule [--quiet] init [--] [<path>...]
   or: git submodule [--quiet] deinit [-f|--force] [--] <path>...
   or: git submodule [--quiet] update [--init] [--remote] [-N|--no-fetch] [-f|--force] [--checkout|--merge|--rebase] [--reference <repository>] [--recursive] [--] [<path>...]
   or: git submodule [--quiet] summary [--cached|--files] [--summary-limit <n>] [commit] [--] [<path>...]
   or: git submodule [--quiet] foreach [--recursive] <command>
   or: git submodule [--quiet] sync [--recursive] [--] [<path>...]
$ git --version
git version 2.7.4
```

earlephilhower, 2019-07-13:
I'm running 18.04, but am surprised that 16.04 has issues.  The actual Makefile portion is pretty simply (if long).

First, how many cores and how much RAM do you have?  I've got a 16 core CPU with 32GB of RAM on my box, but if you're running on a quad core with 8GB or something, you need to tune the jobs way down (out of memory issues).  `-jXX` is normally 2x the number of cores, but if you're memory limited then it needs to be smaller.  As a sanity test, just fire the make off with `-j1` to ensure only one core at a time.

For the failures, check the log.stage.XXX, they record all the command outputs and will be the way to debug things.  There might me some dependencies missing, out of RAM/CPU, etc. shown there.

Niek, 2019-07-13:
Actually I think it's a matter of outdated README. What didn't work for me is:
```
docker run --user $(id -u):$(id -g) --rm -v $(pwd):/workdir earlephilhower/gcc-cross bash -c "cd /workdir; make -j32 GCC=7.2 host=linux GHUSER=earlephilhower GHPASS=x"
```

However this worked fine:
```
docker run --user $(id -u):$(id -g) --rm -v $(pwd):/workdir earlephilhower/gcc-cross bash -c "cd /workdir; make -j32 GCC=7.2 GHUSER=earlephilhower GHPASS=x .stage.LINUX.done"
```
Compiling directly on Ubuntu 16 also works fine as long as the git-core PPA is used, because your submodule commands are only supported on recent git versions.

earlephilhower, 2019-07-14:
Yeah, I updated the readme probably the same time you were trying this (for the gcc7.2 build IIRC). Looks like we're good now.  Expect another merge soon to build pure GNU GCC 9.1.

closed 2019-07-14 by earlephilhower