earlephilhower/esp-quick-toolchain #13: GHUSER and GHTOKEN

kmihaylov, 2020-06-03:
Hello. I'm trying to build ESP Quick Toolchain. After I clone it I issue ```make download```, but it asks me for github credentials. Even if I provide mine, the following error occurs:
```
Makefile:323: recipe for target '.stage.download' failed
```
Isn't it expected that the build use the author's repo?

A bit offtopic, but are some complications expected if I build on FreeBSD?

Thanks!

earlephilhower, 2020-06-03:
It's set up to allow other people to build and upload on their own forks.  You can fork it or set GHUSER to my acct name, and just use garbage to set the GHTOKEN.  The token is only used it you want to upload a release.

As for FreeBSD, I've never tried but if your make and gcc are the GNU versions I don't see why there would be a problem.  I assume a "Linux"  default type but that's just a text string and doesn't change build flow.

If you do find things, feel free to let me know.  Good luck!

kmihaylov, 2020-06-03:
@earlephilhower thank you very much! I believe that it would be good for your explanation to be published within the docs/readme file.

I successfully did ```gmake download```.

Then the build process stopped once or twice w/o any warning, but issuing the command again succeeded to continue.

My last error is Makefile:375, .stage.patch (Error 127). I went through the Makefile and applied the patches manually (bin-*patch):
```
$ patch -p1 < /tmp/esp-quick-toolchain/patches/bin-001-hal.patch
```
and it is successful. Hence I don't know what goes wrong. Some source mentions that Error 127 might be related to missing x64 binary on my system (https://stackoverflow.com/questions/36365752/make-error-127-when-running-trying-to-compile-code#36368057 )

```
$ gmake GCC=9.3 REL=3.0.0 SUBREL=gnu5 -j2
STAGE: .stage.download
STAGE: .clean.binutils-gdb-gnu.git
STAGE: .clean.gcc-gnu.git
HEAD is now at a9d9a104dd Set to 2.32 release version and regenerate configure and pot files
STAGE: .clean.newlib.git
HEAD is now at 671feddab Merge pull request #8 from s-hadinger/patch-2
STAGE: .clean.lx106-hal.git
HEAD is now at e4bcc63 Merge pull request #11 from olaf-mandel/cleanup2
Removing log.stage.patch
STAGE: .clean.mkspiffs.git
HEAD is now at 7fefeac add license file
Removing Makefile.orig
Removing main.cpp.orig
STAGE: .clean.esptool.git
HEAD is now at f80ae31 Update for cross-compiled builds
STAGE: .clean.mklittlefs.git
HEAD is now at c41e51a Set LFS 'c' time using st_mtime, not st_ctime
HEAD is now at 4212a6a3e44 Update ChangeLog and version files for release
Removing cloog
Removing libelf
STAGE: .stage.prepgit
STAGE: .stage.checkout
STAGE: .stage.patch
gmake: *** [Makefile:375: .stage.patch] Error 127
```

Would you please give me a hint on what it might be? Probably some missing library on my FreeBSD machine. Thank you very much!

earlephilhower, 2020-06-03:
There will be hidden `.stage.linux.xxx` files in the build directory.  Take a look inside the failing ones and you should get the exact failure.  All output is piped there from the commands, whenever possible (I normally have -j32 jobs running on my builds and things scroll way too fast for sanity).

If the build stops, then something is funky even if the next stage does start on reinvocation.  I'd look at the silent failure stages to begin with as they might be the root of later failures.

As a first pass tools requirement, you should be able to build a native GCC.  My own system is a root FS that's been around since 2004-ish so it's a little hard to narrow it down on my box. :)

kmihaylov, 2020-06-23:
For the record I needed to install ```autotools``` and ```gmake```. On some places the Makefile should be modified to use gmake.
Later ```gmake``` tries execution of ```x86_64-linux-gnu-ar``` for example, and obviously for FreeBSD it is ```x86_64-portbld-freebsd12.1-gcc-ar9```.

I'm closing this issue now, but I'll update it later, when I successfully build and test the toolchain.
Greetings!

closed 2020-06-23 by kmihaylov