earlephilhower/esp-quick-toolchain #23: GCC -M option produces odd output

mikee47, 2020-12-12:
## Problem

When compiling with -M option to create dependency file on Windows, the output looks like this:

```
alloc.o: \
 s\:/sandboxes/sming-tmp/Sming/Arch/Esp8266/Components/heap/alloc.cpp
```

With existing toolchains (including regular Windows/Linux toolchains) we get this:

```
alloc.o: \
 s:/sandboxes/sming-tmp/Sming/Arch/Esp8266/Components/heap/alloc.cpp
```

So the ':' is being escaped. This causes make to fail on subsequent builds because it doesn't match any defined targets.

Is this intentional behaviour?

## Test makefile

```
TEST_CPP1 = /c/temp/test.cpp
TEST_CPP2 = c:/temp/test.cpp
TEST_CPP3 = c:\temp\test.cpp
TEST_CPP4 = /c:/temp/test.cpp
GCC = $(ESP_HOME)/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++

all:
	echo "" > $(TEST_CPP1)
	$(GCC) $(TEST_CPP1) -M
	$(GCC) $(TEST_CPP2) -M
	$(GCC) $(TEST_CPP3) -M
	$(GCC) $(TEST_CPP4) -M
```

## Output (incorrect) with ESP_HOME set to ESP quick toolchain gnu12/13/20:

```
echo "" > /c/temp/test.cpp
/c/tools/esp-quick-toolchain-gnu20/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++ /c/temp/test.cpp -M
test.o: c\:/temp/test.cpp
/c/tools/esp-quick-toolchain-gnu20/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++ c:/temp/test.cpp -M
test.o: c\:/temp/test.cpp
/c/tools/esp-quick-toolchain-gnu20/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++ c:\temp\test.cpp -M
test.o: c\:\temp\test.cpp
/c/tools/esp-quick-toolchain-gnu20/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++ /c:/temp/test.cpp -M
test.o: c\:/temp/test.cpp
```

## Output (correct) with all other toolchains (espressif, Dec. 2019 quick toolchain, also Host builds):

```
echo "" > /c/temp/test.cpp
c:\tools\esp-quick-toolchain/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++ /c/temp/test.cpp -M
test.o: c:/temp/test.cpp
c:\tools\esp-quick-toolchain/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++ c:/temp/test.cpp -M
test.o: c:/temp/test.cpp
c:\tools\esp-quick-toolchain/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++ c:\temp\test.cpp -M
test.o: c:\temp\test.cpp
c:\tools\esp-quick-toolchain/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++ /c:/temp/test.cpp -M
test.o: c:/temp/test.cpp
```

earlephilhower, 2020-12-12:
This is a GCC 10 "feature" and there's nothing I can go about it, unfortunately.

I can't find the note about it right now, but it's in the GNU-GCC mailing list somewhere and IIRC the solution was to use the corresponding GNU make.

earlephilhower, 2020-12-12:
Looking at it, I think I might have spoke too soon.  If the only spot you're seeing this as a problem is in `-M` then there might be a simple patch to unescape the output since its so limited.  Alternatively, `sed` is an easy workaround.

I'm not sure, however, what `make` will think of it.  Colons are special chars (which is why GCC's escaping them)...

mikee47, 2020-12-12:
Thanks for confirming, didn't think it would be a bug!

earlephilhower, 2020-12-12:
Actually, let's keep this open.  Like I said, I think there is a possibility for removing that colon during depends generation.  It's probably less intricate than some of the other patches we've got already...

earlephilhower, 2020-12-12:
https://github.com/gcc-mirror/gcc/blob/815eb852a2d293331eba2e241a986b8641d4da1f/libcpp/mkdeps.c#L164-L167

`mkdep.c` inside `libcpp` handles the makefile special chars.  It's possible to just disable the `:` escape...

mikee47, 2020-12-12:
'Munge'. A new word for my vocabulary :-)

mikee47, 2020-12-12:
As it's part of GCC 10 the best approach IMHO is to un-escape if required in make scripts. Compilers for other architectures will follow suit and can't modify all of them!

earlephilhower, 2020-12-13:
OK,. fair enough.  If this comes up again we can reopen and the change is already identified above.  Thx!

closed 2020-12-13 by earlephilhower