earlephilhower/esp-quick-toolchain #36: GCC 10.3 broken via backports

earlephilhower, 2022-06-21:
Unfortunately, it seems like #29 or #33 has broken compilation of BearSSL.  I did not try other files, but this is part of the release `install` process to make a PR on the main ESP8266 Arduino repo.  @jjsuwa-sys3175 , any thoughts?

This is built with the toolchain in 3.1.0-gcc10.3
````
Writing new file /home/earle/src/esp-quick-toolchain/arduino/package/package_esp8266com_index.template.json
-------- Installing toolchain
-------- Building and installing BearSSL
*** buffer overflow detected ***: terminated
during RTL pass: final
src/aead/gcm.c: In function 'br_gcm_reset':
src/aead/gcm.c:105:1: internal compiler error: Aborted
  105 | }
      | ^
0x7f34c0e9608f ???
	/build/glibc-SzIz7B/glibc-2.31/signal/../sysdeps/unix/sysv/linux/x86_64/sigaction.c:0
0x7f34c0e9600b __GI_raise
	../sysdeps/unix/sysv/linux/raise.c:51
0x7f34c0e75858 __GI_abort
	/build/glibc-SzIz7B/glibc-2.31/stdlib/abort.c:79
0x7f34c0ee026d __libc_message
	../sysdeps/posix/libc_fatal.c:155
0x7f34c0f82ab9 __GI___fortify_fail
	/build/glibc-SzIz7B/glibc-2.31/debug/fortify_fail.c:26
0x7f34c0f81355 __GI___chk_fail
	/build/glibc-SzIz7B/glibc-2.31/debug/chk_fail.c:28
0x7f34c0ed804e _IO_str_chk_overflow
	/build/glibc-SzIz7B/glibc-2.31/libio/iovsprintf.c:35
0x7f34c0ee5023 __GI__IO_default_xsputn
	/build/glibc-SzIz7B/glibc-2.31/libio/genops.c:399
0x7f34c0ee5023 __GI__IO_default_xsputn
	/build/glibc-SzIz7B/glibc-2.31/libio/genops.c:370
0x7f34c0ec9fe5 __vfprintf_internal
	/build/glibc-SzIz7B/glibc-2.31/stdio-common/vfprintf-internal.c:1719
0x7f34c0ed80f8 __vsprintf_internal
	/build/glibc-SzIz7B/glibc-2.31/libio/iovsprintf.c:95
0x7f34c0f80e4a ___sprintf_chk
	/build/glibc-SzIz7B/glibc-2.31/debug/sprintf_chk.c:40
0x7f34c0e77082 __libc_start_main
	../csu/libc-start.c:308
Please submit a full bug report,
with preprocessed source if appropriate.
Please include the complete backtrace with any bug report.
See <https://gcc.gnu.org/bugs/> for instructions.
````

earlephilhower, 2022-06-21:
Could you please double check that the patch set you have applied matches the one in the repo?  Maybe some of the GCC patches combine cause the problem

The report I posted here is directly from the build of 3.1.0 that I was doing for a new core release, from the makefile...

earlephilhower, 2022-06-21:
Thanks for checking, but I'm still unable to build a working one.

Can you try with the version that was built last night by me (3.1.0).  I manually tried building BearSSL with it (i.e. seeing id there was an issue w/Docker or something) and get the same failure (but in a different spot).

FWIW, I am on Ubuntu 20.04.4 w/all the latest updates.  Maybe there is a system library or setting used that is different between us?

Also I am on git head  8903880f46a97e6b755fa98a4ae6c34483129078 with no diffs.  The patches dir is just like in the repo, no add'l ones:
````
earle@amd:~/src/esp-quick-toolchain/patches/gcc10.3$ ls -l
total 212
-rw-r--r-- 1 earle earle  1873 Jul 15  2019 gcc-ctype-inline-pgmread.patch
-rw-r--r-- 1 earle earle   492 Jul 14  2019 gcc-eh-alloc.patch
-rw-r--r-- 1 earle earle  1336 Jul 15  2019 gcc-enable-mforcel32-eh.patch
-rw-r--r-- 1 earle earle 28124 Jul 15  2019 gcc-exception-what-to-pmem.patch
-rw-r--r-- 1 earle earle   692 Jul 14  2019 gcc-file-shortname.patch
-rw-r--r-- 1 earle earle   643 Dec  7  2019 gcc-func-sect.patch
-rw-rw-r-- 1 earle earle   385 Jul 17  2021 gcc-int32-is-int.patch
-rw-rw-r-- 1 earle earle  8239 May 19  2020 gcc-mforce-l32.patch
-rw-r--r-- 1 earle earle 13167 Jul 15  2019 gcc-regex-what-to-pmem.patch
-rw-r--r-- 1 earle earle  4469 Jul 15  2019 gcc-stdcsupp-safe.patch
-rw-r--r-- 1 earle earle  1995 Jul 15  2019 gcc-unwind-safe.patch
-rw-rw-r-- 1 earle earle  3808 Nov 27  2021 gcc-xtensa-0001-fix-PR-target-98285.patch
-rw-rw-r-- 1 earle earle  2569 Nov 27  2021 gcc-xtensa-0002-rearrange-DI-mode-constant-loading.patch
-rw-rw-r-- 1 earle earle  3901 Nov 27  2021 gcc-xtensa-0003-add-optimizations-for-shift-operations.patch
-rw-rw-r-- 1 earle earle  3871 Nov 27  2021 gcc-xtensa-0004-implement-bswapsi2-bswapdi2-and-helpers.patch
-rw-rw-r-- 1 earle earle  1175 Nov 27  2021 gcc-xtensa-0005-fix-PR-target-102115.patch
-rw-rw-r-- 1 earle earle  1119 Jun 20 17:16 gcc-xtensa-0006-make-trying-to-replace-l32r-with-movi-sll.patch
-rw-rw-r-- 1 earle earle 95827 Jun 20 17:16 gcc-xtensa-0007-Backport-patches-from-upstream-master.patch
````

jjsuwa-sys3175, 2022-06-21:
> Doesn't GCC compile itself after bootstrapping w/whatever native compiler is installed?

of course `apt-get install` manually to install tools/libs that can self-compile the kernel :)

jjsuwa-sys3175, 2022-06-21:
```
0x7f34c0f80e4a ___sprintf_chk
	/build/glibc-SzIz7B/glibc-2.31/debug/sprintf_chk.c:40
```
somehow I understand... it should be a problem with `sprintf()`.
and there are eight places calling `sprintf()`, one of which uses a template obviously larger than the buffer.

```
+(define_insn "bswapsi2_internal"
+  [(set (match_operand:SI 0 "register_operand" "=a,&a")
+	(bswap:SI (match_operand:SI 1 "register_operand" "0,r")))
+   (clobber (match_scratch:SI 2 "=&a,X"))]
+  "!optimize_debug && optimize > 1 && !optimize_size"
+{
+  rtx_insn *prev_insn = prev_nonnote_nondebug_insn (insn);
+  const char *init = "ssai\t8\;";
+  static char result[64];
+  if (prev_insn && NONJUMP_INSN_P (prev_insn))
+    {
+      rtx x = PATTERN (prev_insn);
+      if (GET_CODE (x) == PARALLEL && XVECLEN (x, 0) == 2
+	  && GET_CODE (XVECEXP (x, 0, 0)) == SET
+	  && GET_CODE (XVECEXP (x, 0, 1)) == CLOBBER)
+	{
+	  x = XEXP (XVECEXP (x, 0, 0), 1);
+	  if (GET_CODE (x) == BSWAP && GET_MODE (x) == SImode)
+	    init = "";
+	}
+    }
+  sprintf (result,
+	   (which_alternative == 0)
+	   ? "%s" "srli\t%%2, %%1, 16\;src\t%%2, %%2, %%1\;src\t%%2, %%2, %%2\;src\t%%0, %%1, %%2"
+	   : "%s" "srli\t%%0, %%1, 16\;src\t%%0, %%0, %%1\;src\t%%0, %%0, %%0\;src\t%%0, %%1, %%0",
+	   init);
+  return result;
+}
+   [(set_attr "type"	"arith,arith")
+    (set_attr "mode"	"SI")
+    (set_attr "length"	"15,15")])
```
`result[64]` must be larger.

jjsuwa-sys3175, 2022-06-21:
the testpiece:
```
int test(int a) {
  return __builtin_bswap32(a);
}
```
compiling by [x86_64-linux-gnu.xtensa-lx106-elf-1dc7fdc.220621.tar.gz](https://github.com/earlephilhower/esp-quick-toolchain/releases/download/3.1.0-gcc10.3/x86_64-linux-gnu.xtensa-lx106-elf-1dc7fdc.220621.tar.gz) with `-O2` reproduces buffer-overflow (if with `-Os`, it does not occur due to emitting library call).

earlephilhower, 2022-06-21:
My math seems to show max of 87 chars.  Maybe just double to 128 and be done with it?

jjsuwa-sys3175, 2022-06-21:
> My math seems to show max of 87 chars. Maybe just double to 128 and be done with it?

just like that :)
see https://gcc.gnu.org/pipermail/gcc-patches/2022-June/596990.html

earlephilhower, 2022-06-21:
C is a wonderful language with which to shoot yourself in the foot!  I'll update the patch and rebuild...

earlephilhower, 2022-06-21:
Rebuild passed the BearSSL compile.  I'm starting a new complete cross-compiler run now.

closed 2022-06-21 by earlephilhower

jjsuwa-sys3175, 2022-06-21:
I would like to incorporate the patch for this fix into #37...

PR #39 (2022-06-21): Fix buffer overflow in GCC backport patch
Fixes #36
commit bd0ee836 (2022-06-21): Fix buffer overflow in GCC backport patch

Fixes #36