earlephilhower/esp-quick-toolchain #44: [feature request] save `A0` when `-Og` is enabled

d-a-v, 2023-01-31:
[ per proposal](https://github.com/esp8266/Arduino/pull/8837#issuecomment-1410343314)

Cc: @jjsuwa-sys3175 @mhightower83

jjsuwa-sys3175, 2023-02-01:
The patch should look something like this ([xtensa-Enforce-return-address-saving-when-Og-is-spec.patch](https://github.com/earlephilhower/esp-quick-toolchain/files/10552833/xtensa-Enforce-return-address-saving-when-Og-is-spec.patch)).

jjsuwa-sys3175, 2023-02-21:
The patch has been merged into gcc/master,
[xtensa: Enforce return address saving when -Og is specified](https://github.com/gcc-mirror/gcc/commit/a804419c89db9e1c0b99e772dda6c95095bab0bf)

jjsuwa-sys3175, 2023-02-24:
Of course, this patch will be included in the forthcoming toolchain :)

earlephilhower, 2023-03-12:
@jjsuwa-sys3175 is this in #37?  Can we close this issue (the PR to update to the new toolchain is waiting in the Arduino repo now)

jjsuwa-sys3175, 2023-03-12:
> @jjsuwa-sys3175 is this in #37? Can we close this issue (the PR to update to the new toolchain is waiting in the Arduino repo now)

Yes.

closed 2023-03-12 by earlephilhower