earlephilhower/esp-quick-toolchain #45: `make -j <N>` doesn't work if `download` stage wasn't done yet

slonopotamus, 2023-03-12:
Way to reproduce:

1. Make a fresh esp-quick-toolchain repo clone (or `git clean -ffdx` existing one)
2. `make -j2 make GHUSER=earlephilhower GHTOKEN=unused GCC=10.2`

Expected: build happily runs

Actual:

```
$ make -j2 GHUSER=earlephilhower GHTOKEN=unused GCC=10.2
STAGE: .stage.download
STAGE: .clean.binutils-gdb-gnu.git
/bin/sh: line 1: cd: /hoome/marat/esp-quick-toolchain/repo/binutils-gdb-gnu: No such file or directory
make: *** [Makefile:368: .clean.binutils-gdb-gnu.git] Error 1
make: *** Waiting for unfinished jobs....
```

The reason is that `.clean.%.git` tasks attempt to run before the repositories they try to clean _even start to exist_.

slonopotamus, 2023-03-12:
Correction. `make -j<N>` doesn't even work if I do `make download` as a separate step.

earlephilhower, 2023-03-17:
Download is intended to be a manual stage because of the massive amounts of data it downloads from multiple repos.  This is intended behaviour.

closed 2023-03-17 by earlephilhower