earlephilhower/esp-quick-toolchain #51: .stage.patch error

HamzaHajeir, 2023-09-19:
Hi there, 

Trying to build GCC13.2 using the Espressif's [own fork](https://github.com/espressif/gcc/tree/esp-13_2_0) with introducing those lines into the Makefile:

```lang-make
else ifeq ($(GCC), 12.2)
    ISL           := 0.18
    GCC_BRANCH    := esp-12_2_0
    GCC_PKGREL    := 120200
    GCC_REPO      := https://github.com/espressif/gcc.git
    GCC_DIR       := gcc-gnu
    BINUTILS_BRANCH := binutils-2_32
    BINUTILS_REPO := https://sourceware.org/git/binutils-gdb.git
    BINUTILS_DIR  := binutils-gdb-gnu
else ifeq ($(GCC), 13.2)
    ISL           := 0.18
    GCC_BRANCH    := esp-13_2_0
    GCC_PKGREL    := 130200
    GCC_REPO      := https://github.com/espressif/gcc.git
    GCC_DIR       := gcc-gnu
    BINUTILS_BRANCH := binutils-2_32
    BINUTILS_REPO := https://sourceware.org/git/binutils-gdb.git
    BINUTILS_DIR  := binutils-gdb-gnu
```

And building using the command:
```
 make GCC=13.2 REL=2.5.0 SUBREL=3  GHUSER=earlephilhower GHTOKEN=xxx
```

The error message was 127 pointing to the third line of this script:
```
	for p in $(PATCHDIR)/bin-*.patch; do \
	    test -r "$$p" || continue ; \
	    (cd $(REPODIR)/$(BINUTILS_DIR); echo "---- $$p:"; patch -s -p1 < $$p) ; \
```

The log file (.log.stage.patch) content: 
> ---- /home/hamza/Desktop/esp-quick-toolchain/patches/gcc-001-jump-tables-in-text-section-earlephilhower.patch:
> ---- /home/hamza/Desktop/esp-quick-toolchain/patches/bin-001-hal.patch:
> ---- /home/hamza/Desktop/esp-quick-toolchain/patches/mkspiffs/0.2.0-makefile.patch: 

What might be the problem, and potentially the fix for?
Given that: 
- OS: Ubuntu 22.04.2 LTS

With thanks,
Hamza Hajeir

earlephilhower, 2023-09-19:
Sorry, using a non-GNU fork of GCC is really getting into it on your own.  I would suggest trying the patches one at a time (using the script in the makefile) from the command line and verifying individually what fails.

I have no idea what Espressif has done to their private GNU tree, plus you're using patches not developed for GCC 13 sources, so even if the patches somehow apply there is little expectation of them actually performing as intended.

closed 2023-09-19 by earlephilhower

HamzaHajeir, 2023-09-19:
I see, but in which way you were indicating [utilizing espressif](https://github.com/esp8266/Arduino/discussions/8990#discussioncomment-7022536) one? Perhaps by not making any patch? 

Per the issue, What was missing is the `autoreconf` tool was not installed.

earlephilhower, 2023-09-19:
I thought you were talking about using a plain built binary version.  Building GCC from source is not trivial.  The Docker image mentioned in the README has all the dependencies already set up, but if you want to run native on your own you need a full GNU dev environment.

HamzaHajeir, 2023-09-19:
I see, unfortunately they do provide a patched gcc version only, not built binaries. 

Per the environment, yes correct, but I beleive my problem is bigger than the environment, given that GCC 13.2 isn't supported yet. Just came across an error in .log.stage.gcc1-make with undefined macro (XCHAL_HAVE_MMU). 

This alone perhaps solvable with inserting under the following lines: 
```
	# Dirty-force HAL definition to binutils and gcc
	for ow in $(REPODIR)/$(GCC_DIR)/include/xtensa-config.h $(REPODIR)/$(BINUTILS_DIR)/include/xtensa-config.h; do \
	    ( cat $(REPODIR)/lx106-hal/include/xtensa/config/core-isa.h; \
	      cat $(REPODIR)/lx106-hal/include/xtensa/config/system.h ; \
	      echo '#define XCHAL_HAVE_FP_DIV   0' ; \
              echo '#define XCHAL_HAVE_FP_RECIP 0' ; \
              echo '#define XCHAL_HAVE_FP_SQRT  0' ; \
              echo '#define XCHAL_HAVE_FP_RSQRT 0' ) > $${ow} ; \
```

But still, I'm on my own there...

What about the GCC 11.1.0. Is its support complete there?

Jason2866, 2023-09-19:
Why not building with the espressif forked crosstool-NG?

HamzaHajeir, 2023-09-19:
@Jason2866 

I'm really not aware of it, so it's something esp8266/arduino can rely on (As this is [my purpose](https://github.com/esp8266/Arduino/discussions/8990))?

Jason2866, 2023-09-20:
Never did for esp8266, but the config is there https://github.com/espressif/crosstool-NG/tree/esp-1.25.x/samples/xtensa-esp-elf

HamzaHajeir, 2023-09-23:
> Never did for esp8266, but the config is there https://github.com/espressif/crosstool-NG/tree/esp-1.25.x/samples/xtensa-esp-elf

Have downloaded the release, but the folder xtensa-esp-elf seems missing from each the Source Code and `crosstool-NG-esp-13.2.0_20230919-src.tar.xz` releases!

Would that prevent us from building for the 8266?

Jason2866, 2023-09-23:
No, the just do not build and release for esp8266. The Config setup for esp8266 is in the CROSSTOOL-NG repo. With the current changes it does not work at all. The new added submodules are not available.

HamzaHajeir, 2023-09-23:
I did notice their submodules are not publicly released so thought to find them within the source code released version.

But it seems to me they are in preparation for esp8266 support, correct?