earlephilhower/esp-quick-toolchain #53: Python crashing

waltje, 2024-05-15:
What is the origin of the python dist in the releases?

It crashes on start, not good...

waltje, 2024-05-15:
Answer: the distributed ZIP file contains the *known to crash* release of Python.  The "correct" source dist is at https://www.python.org/ftp/python/3.7.2/[python-3.7.2.post1-embed-amd64.zip](https://www.python.org/ftp/python/3.7.2/python-3.7.2.post1-embed-amd64.zip), which indeed works.

earlephilhower, 2024-05-15:
That file has been the same for many years now and it hasn't been reported by the thousands of downloaders of it as far as I can see.  Can you be more specific what the issue is, where it's documented, etc.?

Also, are you saying you can only run the 64b version?  We ship a 32b version since we support older Win32 users still.  FWIW I have only run that 32b executable on 64b Windows without incident over Win 10 and Win 11 (well to be fair...with Win 11 who knows if/when they update and break something).  IIRC, it was actually picked up at a much older release than 1yr ago...

There's no particular love for that specific release, though, so if it's just "that version has a rare bug" we can see about bumping to a newer standalone 32b version if there's ever another toolchain release.

waltje, 2024-05-15:
Yes, I noticed it had been the same for years.

I got the following error after compilation, when it was going to start the upload using upload.py:

--------------------------
`Fatal Python error: initfsencoding: unable to load the file system codec
ModuleNotFoundError: No module named 'encodings'

Current thread 0x00003578 (most recent call first):

exit status 3
------------------------------

which you can find MANY references for using Google.  Most "fixes" to that problem do not actually fix anything.

I then tried to execute that version of Python directly, in a command window.  Boom, same thing, same crash. So,
it definitely has to do with Python itself.

Downloaded the same distribution from python.org and... that *does* work. Wha?

Reading up on the site shows a little note on the download page, warning Windows uses that a fatal bug was fixed, and that the downloadable has been updated. They also include a link to the bug-track discussion, which indeed shows why this happens, that they fixed it, and that they decided to not issue a new release, but instead just replaced the downloable:

-------------------------------
UPDATED 2019-01-09: An issue was discovered in the embeddable packages for Windows and updated download files have been provided for the Windows x86-64 embeddable zip file and the Windows x86 embeddable zip file and their GPG signatures. No other download was affected. See https://bugs.python.org/issue35596 for more information.
-------------------------------

Your embedded distr must be from before that date.

After zapping the 'bad' distr and replacing it with the one from python.org, all is well, although I did have to copy python.exe to python3.exe, as that was expected by the rest of the code.

The uploader now works.

One more glitch: Not sure why, bit it failed executing Python because 

    tools.esptool.cmd={runtime.tools.python3.path}/python3

in platform.txt

does not expand, the path expands to __REMOVE__/python3.

For now, I have explicitly set that to the correct path, and changed 'python3' to 'python' while I was at it since the python.org distr uses that name.

Let me know if more info is needed, but I do think a new release is warranted.

earlephilhower, 2024-05-15:
That `encodings` issue is actually an IDE installation issue and not a Python actual bug.  For some reason the IDE doesn't properly expand the full ZIP file and it ends up missing/corrupted occasionally when you do an update.  The fix has been a Boards Manager uninstall followed by reinstall.

waltje, 2024-05-15:
Yes, but the zip handler in Python 3.7 was updated as a result.

waltje, 2024-05-15:
Update: the latter might be because the original package was per Arduino 1.5 specs, I am using 1.8.something. Maybe they changed those defs and the python3 path is no longer there.

waltje, 2024-05-15:
Also, note that the bug shows even if you execute python.exe directly, or, at least, onsome systems/platforms.  That's why python.org fixed it on their end ;-)
--fred

earlephilhower, 2024-05-15:
The IDE doesn't use the Python extractor, they're Java or TypeScript based and use those native libraries.  It's really just the upgrade not writing the files properly, for some reason.  After the busted installation, you can run it manually or from the IDE and will get the same issue since files are physically missing.

There's been a bug at the Arduino repo about it for a while, and the Pico equivalent debug was tracked here:  https://github.com/earlephilhower/arduino-pico/issues/976

waltje, 2024-05-15:
Yes, I did see the Pico one.

 
 

There's been a bug at the Arduino repo about it for a while, and the Pico equivalent debug was tracked here: earlephilhower/arduino-pico#976